microbit-foundation/micropython-microbit-v2 #83: Pin drive strength is not the same V1 and V2

DaveAtKitronik, 2021-07-26:
Reported by a customer - the ZIP64 does not function correctly with microPython on a V2 microbit.
This is the same as a resolved issue in makecode.  (https://github.com/microsoft/pxt-microbit/issues/3825).

We added: 
#if MICROBIT_CODAL
      uBit.io.P0.setHighDrive(true);
 #endif
in a c++ shim to fix it there. Not sure where a fix would be applied in microPython - to the neopixel code (baked in), or to just expose the ability to change the pin drive  strength from the python code

dpgeorge, 2021-09-20:
I suggest to change the pin drive strength by writing directly to nRF GPIO registers, using `machine.mem32`.

dpgeorge, 2021-09-20:
The equivalent C code to set high drive is:
```c
    uint32_t s = PORT->PIN_CNF[PIN] & 0xfffff8ff;
    s |= 3 << 8;
    PORT->PIN_CNF[PIN] = s;
```

dpgeorge, 2021-09-20:
Here's a function that should be able to set the high drive on a pin (I've not tested it though):
```python
def set_high_drive(port_id, pin_id):
    addr = 0x5000_0000 + port_id * 0x300 + 0x700 + pin_id * 4
    machine.mem32[addr] = (machine.mem32[addr] & 0xffff_f8ff) | 3 << 8
```
For P0.00 to P0.31 use `port_id=0`, for P1.00 to P1.15 use `port_id=1`.

kevinjwalters, 2025-03-08:
I may have duped this with https://github.com/microbit-foundation/micropython-microbit-v2/issues/216