microbit-foundation/micropython-microbit-v2 #87: radio.RATE_250KBIT in V1 but not V2

microbit-matt-hillsdon, 2021-09-08:
`radio.RATE_250KBIT` is not defined in MicroPython for V2 but is for V1.

It's commented out in the source: https://github.com/microbit-foundation/micropython-microbit-v2/blame/eba8995843ebc7246765b364710543c9ffee344a/src/codal_port/modradio.c#L269 (and elsewhere).

It's documented in the V2 docs `radio.config` `data_rate` parameter: https://microbit-micropython.readthedocs.io/en/v2-docs/radio.html#radio.config

It looks like the relevant MODE is deprecated for nRF52. [This DevZone response](https://devzone.nordicsemi.com/f/nordic-q-a/78469/250-kbit-s-nordic-proprietary-radio-mode-on-nrf52840) that Carlos found suggests it is not tested.

If it's not reasonable to support it for V2 then we should mention this in the docs for the parameter. Happy to create a PR for that but raising this for discussion first.

dpgeorge, 2021-09-20:
In a52cd59732e49005afd4a9e805e1e9971ae29574 I made it so the user can pass in a value of "2" to select 250Kbit, if they really need it to communicate with a micro:bit v1.  But otherwise they shouldn't use it and so the constant `radio.RATE_250KBIT` is unavailable.

microbit-carlos, 2022-03-25:
I've tested this with a micro:bit V1.3 and V2.2 and it works!

```python 
from microbit import *
import radio

rate = radio.RATE_250KBIT if hasattr(radio, "RATE_250KBIT") else 2
radio.config(data_rate=rate)
radio.on()

while True:
    if button_a.is_pressed():
        display.show(Image.SURPRISED)
        radio.send("hello")
    else:
        received_data = radio.receive()
        if received_data:
            display.show(received_data)
        else:
            display.show(Image.HAPPY)
    sleep(500)
```

I've also raised this PR to remove `radio.RATE_250KBIT` from the docs: https://github.com/bbcmicrobit/micropython/pull/731

closed 2022-03-25 by microbit-carlos