microbit-foundation/micropython-microbit-v2 #89: Undefined symbols for arm64 - Issue building micropython v2

scath999, 2021-11-23:
Have tried to build this project per the instructions for building v2 [here](https://microbit-micropython.readthedocs.io/en/v2-docs/devguide/flashfirmware.html#id2).

End up with the following link error:

```
LINK mpy-cross
Undefined symbols for architecture arm64:
  "_nlr_push", referenced from:
      _mp_iternext in runtime.o
      _it_iternext in objgetitemiter.o
      _main_ in main.o
     (maybe you meant: _nlr_push_tail)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [mpy-cross] Error 1
```

Any help?

microbit-carlos, 2021-11-23:
What OS and architecture are you using? Is it Linux or macOS in arm64?

scath999, 2021-11-23:
My bad.
Mac M1 - Monterey

microbit-carlos, 2021-11-23:
Thanks @scath999, looks like it might be this issue: https://github.com/micropython/micropython/issues/7232

You could apply the patches listed there locally, or maybe try if you can install x86 gcc with rosetta and that might work?

scath999, 2021-11-24:
@microbit-carlos Much appreciated. That did fix the one issue - have now found others, but that's my boat to row.

Cheers!

closed 2021-11-24 by scath999