microbit-foundation/micropython-microbit-v2 #91: datalog: The "header" timestamp should be enable even without calling log.set_labels()

microbit-carlos, 2022-03-17:
The `log.set_labels()` feature has a default value for the `timestamp` argument, currently `log.MILLISECONDS`. 

When the `set_labels()` function is called this default unit for the timestamp is correctly applied. However, we also want to make sure that the timestamp is set with the default value even if `set_labels()` isn't called.

So this small programme should log the temperature with a timestamp, which it currently doesn't do:
```python
import log

log.add(temp=temperature())
```

Related:
-

dpgeorge, 2022-04-11:
Is it correct to just call `uBit.log.setTimeStamp(TimeStampFormat::Seconds)` when the device starts up?  Will that add a new header each time the micro:bit is reset?

microbit-carlos, 2022-04-11:
> Is it correct to just call `uBit.log.setTimeStamp(TimeStampFormat::Seconds)` when the device starts up? Will that add a new header each time the micro:bit is reset?

I'm not sure, @JohnVidler do you know the answer to this question?

JohnVidler, 2022-04-12:
I've had a dig and it looks like if you call `setTimeStamp(...)` it'll mark the headers as dirty, and the micro:bit will write them to the log when the next actual bit of real data is written; so as the initial value is `None`, if you set it to anything else you'll get a new header row as soon as you write new data.

microbit-carlos, 2022-04-22:
So, if MicroPython calls `setTimeStamp(...)` on startup, but it never adds new data it doesn't add any data to the log file, is that right?

Do you know if it also trigger the MY_DATA.HTM file to appear?

microbit-carlos, 2022-04-22:
Btw, the MakeCode source code can be found here:
- C++ side https://github.com/microsoft/pxt-microbit/tree/v4.1.17/libs/flashlog
- Extension https://github.com/microsoft/pxt-microbit/tree/v4.1.17/libs/datalogger

microbit-carlos, 2022-07-01:
Looks like a CODAL programme only doing this will still trigger the MY_DATA.HTM file:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uBit.init();

    uBit.log.setTimeStamp(TimeStampFormat::Seconds);

    while (true) {
        uBit.display.print("hello world!");
    }
}
```

So the only way to avoid this might be to start each log module function with a "if not init: initialise()" as it's currently done in MakeCode. 

Another option might be to run that on import? Although import side effects are generally frown upon, but at least we are not constantly running an if at each function call.

Do we have any other options?

dpgeorge, 2022-08-21:
> Another option might be to run that on import? Although import side effects are generally frown upon, but at least we are not constantly running an if at each function call.

Considering that `import radio` now turns the radio on, I think it's fair enough that `import log` sets the default timestamp.

dpgeorge, 2022-08-21:
Doing `import log` now sets the default timestamp format to seconds, see 6c511644812a0b15671a7c846d897afeadf3072c