microbit-foundation/micropython-microbit-v2 #93: DHT11 / DHT22 do not work with microbit v2

leomlr, 2022-03-18:
Hi @microbit-carlos @dpgeorge 

Many of our users from Vittasciennce.com/microbit/ complain about the poor performance of the DHT11 and DHT22 sensors with the micro:bit v2 board and MicroPython. Is there anything we can do to make it work with this version.

I note that the DHT11 sensor works correctly with the micro:bit v1 and the one wire protocol. However, it requires a lot of memory.

Waiting for your help. Thanks very much !

microbit-carlos, 2022-03-18:
Hi @leomlr,

Do you have any specifics on how it is different in micro:bit V2 and MicroPython? In what way is the performance poor?

If you could provide a simple way to replicate this difference with a V1 and V2 that would also be useful.

leomlr, 2022-03-21:
For the following program, the behaviour of micro:bit v1 and v2 is different.
```
from microbit import *
from dht11 import DHT11
import time

dht11_1 = DHT11(pin1)

while True:
  print(dht11_1.getData(d=1))
  print(dht11_1.getData(d=2))
  time.sleep(1)
```

The v1 is working fine and return the temperature and moisture correctly.
For the v2, the measure do not work. The sensor return this error:
[image]

It seems the protocol of linked library [dht11](https://github.com/vittascience/microbit-libraries) do not work in same way for the 2 versions of micro:bit.

microbit-carlos, 2022-03-21:
It looks like there is quite a bit of assembly in that library. Have you checked if the assembly is compatible with the nRF52833 (Cortex-M4f) of the micro:bit V2 and that any pins or register you might be accessing are compatible?

leomlr, 2022-03-24:
Thanks for replies. The assembly is quite different with nRF52833 of micro:bit v2. I try a fix by using work of sndmdk https://github.com/sndmtk/microbit-v2-dht11 

Effectively, the DELAY loop needs to be increased by changing r7 to 0xb4 (x4 compared to micro:bit v1).
GPIO pins are different by checking CPU at https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fgpio.html&cp=4_2_0_19&anchor=concept_zyt_tcb_lr

The library is not as stable as with microbit v1. However, measurements are valid approximately every 20 seconds with microbit v2. Unfortunately, my skills are limited in assembler. This library could be greatly improved.

microbit-carlos, 2022-03-25:
> Effectively, the DELAY loop needs to be increased by changing r7 to 0xb4 (x4 compared to micro:bit v1).

V1 runs at 16MHz and V2 at 64Mhz, so the 4 multiplier needed makes sense.

> The library is not as stable as with microbit v1. However, measurements are valid approximately every 20 seconds with microbit v2. 

Could you expand a little bit more what's the different between V1 and V2 after those changes?

dpgeorge, 2022-04-11:
It looks like you're using `utime.sleep_ms(...)` with interrupts disabled.  That probably won't work well.  You can try instead to use `utime.sleep_us(...)`.

In upstream MicroPython there is a DHT driver which uses C code for the low-level bit operations.  That works well on a variety of MCUs.  Maybe we can bring it into microbit?

rhubarbdog, 2022-12-19:
have you tried the code in repository https://github.com/rhubarbdog/microbit-dht11 which i think your repository is cloned from?
I also had issues with version 2 microbits. i now have more microbits, but only one sensor so when i do further testing i may discover what the problem is.

rhubarbdog, 2022-12-19:
@leomir the dht11 should only be sampled every 2 seconds as in the pdf

closed 2024-02-24 by leomlr