microbit-foundation/micropython-microbit-v2 #96: Random freezes when playing tones on Microbit V2

pakal, 2022-05-06:
When playing tones multiples times, by button press or edge events on Pins (tried P1 or P2), the board freezes after a few or many attempts.

Demo :
https://makecode.microbit.org/_FRjWbe2pJMbX

It sounds like https://github.com/microbit-foundation/micropython-microbit-v2/issues/73, but Makecode uses a recent enough CODAL version however (codal-microbit-v2 runtime version:  [v0.2.32](https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.32))

I've tried with a without an external speaker board, different tones, differnts sounds durations, to no avail. Pleying the SAME tone multiple times shortly seems to freeze the board more easily though.

microbit-carlos, 2022-05-09:
Hi @pakal,

Thanks for the report! But just to let you know that this repository is for MicroPython, which can be used via the Python Editor: https://python.microbit.org/v/beta

If you have issues with MakeCode (Blocks, TypeScript, or Python), the place to raise that would the pxt-microbit repository: https://github.com/microsoft/pxt-microbit/issues

pakal, 2022-05-10:
My bad, I opened a proper ticket on https://github.com/microsoft/pxt-microbit/issues/4627

I could not reproduce this bug with micropython douns utilities, which is good news.

closed 2022-05-10 by pakal