microbit-foundation/micropython-microbit-v2 #107: microphone LED is on at boot since CODAL 0.2.35

jaustin, 2022-08-01:
@dpgeorge tagging here.

Recent CODAL versions bring the microphone up automatically with that audio pipeline.

Therefore, I think this line:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/4d71da807ee299db979cc3c39cfb16d63c7fef8a/src/codal_app/main.cpp#L71

... will cause the audio pipeline to to come up, which causes the mic to turn on

Some options would be
1) defer the message bus listening until the first time that the user requests some form of audio event (of course, this means we'll likely miss the first time they ask for one) - eg in `microbit_hal_microphone_init`
2) explicitly turn off the microphone after that initialisation: https://github.com/lancaster-university/codal-microbit-v2/blob/b587d9c26f5e93a2054ba8b36a6cdcc27b234419/source/MicroBitAudio.cpp#L113

In looking over the code @JohnVidler  and I noticed that https://github.com/microbit-foundation/micropython-microbit-v2/blob/e0f3e60c5e7755eb71c789b3a765d271320b2e17/src/codal_app/microbithal_microphone.cpp#L39 is perhaps doing a lot more than it should these days, as we have a LevelDetector in uBit now (which gives us consistent parameters across MakeCode and MicroPython, by design, but I guess right now not in reality).

I'll let @JohnVidler chime in on what the right approach is here!

JohnVidler, 2022-08-08:
The python codebase seems to be spinning up its own independent pipeline for audio, but really should be using the uBit object defs, as the parameters there have been extensively tweaked for best performance, etc.

See https://github.com/microbit-foundation/micropython-microbit-v2/blob/318595a5e545e0176e106ff0db8562ca84fd622e/src/codal_app/microbithal_microphone.cpp#L39 and friends.

Is there a particular reason for the python firmware to be doing this itself that I'm not aware of?

microbit-carlos, 2022-08-10:
@dpgeorge do you have some time to look into this before the end of this work week? @JohnVidler might be away on holiday next week and if we need anything done in CODAL we should aim to do it before that.

dpgeorge, 2022-08-15:
> Therefore, I think this line:
> ...
> will cause the audio pipeline to to come up, which causes the mic to turn on

Yes you are correct.  Postponing that registration of event until the microphone is used fixes the problem (the LED is not on, and power consumption is not increased, until a microphone method is called).

> The python codebase seems to be spinning up its own independent pipeline for audio, but really should be using the uBit object def

I did not realise the `uBit` object had all the parts needed for audio (and maybe I wrote that `microbit_hal_microphone_init` function before `uBit` supported microphone level detection??).

I tried removing all that custom initialisation code and the sound level detection still works.

dpgeorge, 2022-08-15:
Fixed by 4f369bbc2e43dcff1a33c7f01f9e8080286e6c48

closed 2022-08-15 by dpgeorge