microbit-foundation/micropython-microbit-v2 #111: Calling spi.init() twice causes a panic with error code 090

microbit-carlos, 2022-08-22:
Panic 090 is the `DEVICE_HARDWARE_CONFIGURATION_ERROR `.

Likely hitting this CODAL panic:
https://github.com/lancaster-university/codal-nrf52/blob/ac01d74dc13236beddde78a5dab23c9e41b42c23/source/NRF52SPI.cpp#L55

But technically MicroPython is deleting the CODAL SPI instance before recreating it, which you'd guess it would free up the peripheral:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/f3db1e6215cd747e8a09f80cb7c35cf9d8f9cad5/src/codal_app/microbithal.cpp#L245-L259

This might be an issue in the CODAL side, in which case we can move this issue there.

microbit-carlos, 2022-08-22:
Right, I don't think there is a destructor in `NRF52SPI.cpp` and I don't see anywhere else where `free_alloc_peri()` might be called for the SPI, so it looks like this is something that could be added to CODAL: https://github.com/lancaster-university/codal-microbit-v2/issues/219

Alternatively, if CODAL provided a way to change the frequency, mode and pins that could be used instead.

@dpgeorge do you have any preference?

dpgeorge, 2022-08-30:
> Alternatively, if CODAL provided a way to change the frequency, mode and pins that could be used instead

CODAL already provides a way to change the frequency and mode.  So all that's needed is a way to change the pins.  That would be better than destructing and recreating the NRF52SPI object (although I still think it would be good to fix CODAL so that it does call `free_alloc_peri()` in the destructor).

microbit-carlos, 2022-09-22:
The CODAL changes required to reassign pins spawn several repos and files, so to reduce risk for the v2.1.0 final release we'll do this for v2.2.

dpgeorge, 2023-02-13:
CODAL was updated to 0.2.48 and SPI fixed in 3af0abd392fd2e325c5e824a2ffb508088ac849a

microbit-carlos, 2024-05-02:
Thanks Damien!

closed 2024-05-02 by microbit-carlos