microbit-foundation/micropython-microbit-v2 #114: `microphone.sound_level()` values are too low

microbit-carlos, 2022-08-24:
Running this programme with the latest CI build and the max value it prints is around 7:

```python
from microbit import *

while True:
    print(microphone.sound_level())
    sleep(10)
```

Likely from 4f369bbc2e43dcff1a33c7f01f9e8080286e6c48, although I haven't tested the parent commit to confirm if that's the cause.
For reference, the microphone min and max values used by MakeCode for scaling are bit a different:
- https://github.com/microsoft/pxt-microbit/blob/d80514bda223f729d452c1e51047f3272a2317fb/libs/microphone/microphone.cpp
- https://github.com/lancaster-university/codal-microbit-v2/blob/864e7b87d2968c91c84e8eb204836b67accabda4/model/MicroBit.h#L323-L325

@dpgeorge could you have a look at this one once you get a chance? I've added it to the 2.1 milestone as well.

dpgeorge, 2022-08-25:
It's not due to the commit 4f369bb, I tested prior to that and it's the same output (max I could get was 5).

dpgeorge, 2022-08-25:
Using `uBit.audio.level->getValue()` the return values are in the range 0 to 480 in steps of 8 (according to my testing).

Using `uBit.audio.levelSPL->getValue()`, the return values are in the range 52 to 86 in steps of 1.

These different values don't have a 1-1 mapping, the `levelSPL` seems to be more noisy/sensitive.

Which one should we use in MicroPython?  How should we scale the range?

microbit-carlos, 2022-08-25:
We'll have to check this with @JohnVidler & @finneyj when they are back next week, but in the meantime let's use the levelSPL with the min and max ranges you've found on your testing.

MakeCoded uses a larger max rate, but that is likely the cause of this:
- https://github.com/lancaster-university/codal-microbit-v2/issues/225

microbit-carlos, 2022-08-26:
We had a chat with @finneyj and for MicroPython and MakeCode we'd like to use `levelSPL`.
For consistency CODAL should provide the min and max values, and it's possible CODAL we'll need to recalibrate the current internal calibration, but that will come in a future CODAL release.

In the time being, let's use `levelSPL` with the min and max values you've found via testing @dpgeorge.

dpgeorge, 2022-08-30:
Commit 904731430b47e745fd15e8b0565d8ad44100809c switched to use `levelSPL` with a range of 52-86.

closed 2022-08-30 by dpgeorge