microbit-foundation/micropython-microbit-v2 #115: Missing hex for v2.1.0-beta.1

aivarannamaa, 2022-08-31:
Congratulations for the new release!

The hex-file seems to be missing from https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.1.0-beta.1. Is following soon?

microbit-carlos, 2022-08-31:
Ah, the same is true with the V1 beta release:
- https://github.com/bbcmicrobit/micropython/releases/tag/v1.1.0-beta.1

Do you load the hex file from Thonny? Do you keep both hex files individually or use a Universal Hex?

closed 2022-08-31 by microbit-carlos

aivarannamaa, 2022-08-31:
> Do you load the hex file from Thonny? Do you keep both hex files individually or use a Universal Hex?

Yes, there is an [online json](https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-daplink.json) that Thonny consults each time before downloading the firmware. V1 and V2 have separate entries, so for Thonny it doesn't really matter if there is an universal hex available.

https://github.com/bbcmicrobit/micropython/releases/download/v1.1.0-beta.1/microbit-micropython-v1.1.0-beta.1.hex and https://github.com/microbit-foundation/micropython-microbit-v2/releases/download/v2.1.0-beta.1/micropython-microbit-v2.1.0-beta.1.hex are both specific to micro:bit version, right?

aivarannamaa, 2022-08-31:
I'm going to add _v1.1.0-beta.1_ and _v2.1.0-beta.1_ as download options to this [online json](https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-daplink.json) I mentioned before. The first download option gets preselected when the user opens the flashing dialog.

Would you recommend setting the new versions as default / preselected options or leave them as secondary options for now?

microbit-carlos, 2022-08-31:
We'll be adding the beta versions of MicroPython to the beta version of the Python editor, but not the "live" one: 
https://python.microbit.org/v/beta

So we consider them stable, but some of the new V2 API in this release (datalogging, SoundEffects, power, etc) might be tweaked before the final release, so programmes created with the beta API might not work in the final release.

Apart from that nothing should be broken, except for a couple of outstanding things in the new power module, but that's the thing that might be tweaked the most before release (or if we find any blockers there, we might pull the power module from the 2.1.0 release and only have it available in betas).

So I guess it's up to you how stable/experimental you'd like the default.

aivarannamaa, 2022-09-02:
I added the new betas as secondary download options for now.

microbit-carlos, 2022-09-02:
Sounds good, thanks!
Shouldn't take too long for the final releases to be out 👍