microbit-foundation/micropython-microbit-v2 #117: Should a Python programme "end" when there is a `run_every` scheduled?

microbit-carlos, 2022-09-05:
If we have a simple programme like this, the scheduled `log_temp` function never runs, as the programme ends:

```python
from microbit import *
import log

@run_every(s=5)
def log_temp():
    log.add(temp=temperature())
```

So we need to add a busy infinite loop to ensure the programme continues running forever:

```python
from microbit import *
import log

# Set the timer to log data every 5 seconds
@run_every(s=5)
def log_temp():
    log.add(temp=temperature())

while True:
    sleep(100)   # Needed so the programme doesn't end
```

Should we implicitly add an infinite loop if there are scheduled function? The answer might be "no", but it's worth considering.
Assuming Ctrl+C stops everything (which I believe it does), would there be any disadvantage to having that?

dpgeorge, 2022-09-07:
> Should we implicitly add an infinite loop if there are scheduled functions?

I don't think we should.  The user code may sleep for a finite time (eg 5 seconds) and so they would get confused if after that it also slept forever.

Also, the user may add an infinite loop themselves, and then expect ctrl-C to stop that loop and drop to the REPL.  They would be surprised to find out they need to use ctrl-C twice to get to the REPL.

microbit-carlos, 2022-09-15:
Agreed, all good reasons not to implement this 👍

closed 2022-09-15 by microbit-carlos