microbit-foundation/micropython-microbit-v2 #118: Update `power.deep_sleep` to sleep until the next `run_every` and then go back to sleep

microbit-carlos, 2022-09-05:
>As discussed elsewhere, my suggestion is that `run_every=True` does not end the `deep_sleep()` call, but rather wakes briefly to execute the run every and then resumes the deep sleep.  The deep sleep only ends when the given timeout expires, or a `wake_on` event occurs.  In particular, `power.deep_sleep(run_every=True)` will never return.  (And maybe we can then make `run_every=True` the default.)

_Originally posted by @dpgeorge in https://github.com/bbcmicrobit/micropython/issues/754#issuecomment-1232802628_

microbit-carlos, 2022-09-12:
- https://github.com/bbcmicrobit/micropython/pull/765

dpgeorge, 2022-09-21:
Implemented by cd8bbb8d089353d6d6f06be25f628b003a1d3330

closed 2022-09-21 by dpgeorge