microbit-foundation/micropython-microbit-v2 #120: `SoundEffect` with frequencies above 10 KHz don't make a sound

microbit-carlos, 2022-09-06:
I'm guessing the frequency string format only takes 4 digist?

This works:
```
audio.play(audio.SoundEffect(
    freq_start=9999,
    freq_end=9999,
    duration=300,
    vol_start=255,
    vol_end=255,
))
```

But this doesn't make a sound:
```python
audio.play(audio.SoundEffect(
    freq_start=10000,
    freq_end=10000,
    duration=300,
    vol_start=255,
    vol_end=255,
))
```

Should we throw an exception or clamp for values >10K?

microbit-carlos, 2022-09-06:
I haven't tried but I guess the same would be true for volume and duration.

dpgeorge, 2022-09-07:
> I'm guessing the frequency string format only takes 4 digits?

Yes that's correct.  Most of the numbers (volume, frequency, duration) are limited to 4 digits.

> Should we throw an exception or clamp for values >10K?

Implementing either of those is easy.  Probably best to raise an exception.

microbit-carlos, 2022-09-20:
> Probably best to raise an exception.

Sounds good 👍

dpgeorge, 2022-09-21:
Fixed by 0b7552643cb0ab78e511b0772beca2fe3b970833

closed 2022-09-21 by dpgeorge