microbit-foundation/micropython-microbit-v2 #122: Update `microphone.sound_level()` to use new CODAL features

microbit-carlos, 2022-09-08:
A continuation from the discussion in:
- https://github.com/microbit-foundation/micropython-microbit-v2/issues/114

The CODAL updates have been pushed, but a new tag is not released yet (that'll take a few days more as other bug fixes are implemented):
- https://github.com/lancaster-university/codal-microbit-v2/issues/225#issuecomment-1241023894

microbit-carlos, 2022-09-12:
@dpgeorge CODAL v0.2.42 has been released:
- https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.42
- https://github.com/lancaster-university/codal-microbit-v2/compare/v0.2.41...v0.2.42

dpgeorge, 2022-09-21:
New CODAL interface is used by c5ab62716a72147893f8e3672f3f5fb177925e1f

closed 2022-09-21 by dpgeorge

microbit-carlos, 2022-09-22:
As a heads up, I've raised this issue in the CODAL repo about this version of the audio.level:
- https://github.com/lancaster-university/codal-microbit-v2/issues/236