microbit-foundation/micropython-microbit-v2 #124: Playing a `microbit.Sound` or `audio.SoundEffect` crashes beta.2

microbit-carlos, 2022-09-16:
To replicate:

```python
from microbit import *

audio.play(audio.SoundEffect())
display.show("A")
```

```python
from microbit import *

audio.play(Sound.HAPPY)
display.show("A")
```

In both cases the sound will play, but it will get stuck playing the last tone and play that indefinitely, never reaching the line to display "A".

Looks like it was introduced by this CODAL commit: https://github.com/lancaster-university/codal-microbit-v2/commit/31095d5d9f3c1ffc524c2777173d10fe7ee9ab25

More info in:
- https://github.com/lancaster-university/codal-microbit-v2/issues/228

microbit-carlos, 2022-09-16:
To have a working MicroPython hex for the Python Editor, we'll see tomorrow to maybe create a new CODAL tag with the commit reverted, or maybe go back in MicroPython to CODAL v0.2.40.

dpgeorge, 2022-09-16:
This will block #118, #119, #122 (which require features from CODAL 0.2.40).

microbit-carlos, 2022-09-16:
Okay, we've got a hotfix patch to resolve this and I'll release beta.3 right away.

Thanks @JohnVidler!

closed 2022-09-16 by microbit-carlos