microbit-foundation/micropython-microbit-v2 #126: `power.deep_sleep()` without wake up sources does not go to sleep

microbit-carlos, 2022-09-20:
This might be related to this CODAL issue:
- https://github.com/lancaster-university/codal-microbit-v2/issues/203

However, based on the latest conversations with @JohnVidler we were thinking the issue might be in the MakeCode extension instead of CODAL.

However, in MicroPython executing `power.deep_sleep()` doesn't go to sleep either. Having a brief look the MicroPython code I can't see an obvious reason why it might not go to sleep if it's supported within CODAL itself:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/2233d79b8ab15bb769adaa44d53874ce80ebd3eb/src/codal_port/modpower.c#L54-L109
https://github.com/microbit-foundation/micropython-microbit-v2/blob/6ea476f3ddb4eb7aba13ca264fab9d15abaae3ca/src/codal_app/microbithal.cpp#L130-L136

dpgeorge, 2022-09-21:
I think this is a "feature" of the CODAL: if there are no wake up sources configured, and no timeout, then the `deepSleep()` call returns immediately (presumably to prevent infinite waits that cannot be interrupted, requiring a reflash of the microbit to recover).

dpgeorge, 2022-09-21:
We could fix this on the MicroPython side by just calling the CODAL `deepSleep(ms, true)` with a large timeout, and keep calling it in a loop forever.

microbit-carlos, 2022-09-21:
This has been changed/fixed in the CODAL side in this PR, so we can just wait for the next tag to get the fix:
- https://github.com/lancaster-university/codal-microbit-v2/pull/229

microbit-carlos, 2022-09-22:
This is resolved in CODAL v0.2.43, we can close it when https://github.com/microbit-foundation/micropython-microbit-v2/pull/132 is merged, or a different commit updates CODAL in the main branch.

dpgeorge, 2022-09-26:
CODAL v0.2.43 is updated here.

closed 2022-09-26 by dpgeorge