microbit-foundation/micropython-microbit-v2 #131: Scheduled calls from `run_every` are queued during sleep when `deep_sleep(run_every=False)`

microbit-carlos, 2022-09-21:
So a programme like this one, which purposely turns off the `run_every` events during sleep, will wake up and run the `print_time()` function 5 times in a row:
```python
from microbit import *
import power

@run_every(s=3)
def print_time():
    print(running_time())
    display.show([Image.HEART, Image.HEART_SMALL])

while True:
    power.deep_sleep(15_000, run_every=False)
```
The heart animation can be seen 5 times, and this is printed to serial:
```
15182
15985
16789
17593
18397
```

I think in the occasions when `deep_sleep` is set up with `run_every=False`, we wouldn't expect the events to still be queued and all of them run when it wakes up, as the board might be put to sleep for long periods of time (for example, a data logging app that is meant to be deployed somewhere and start/stop logging when a button is pressed).

microbit-carlos, 2022-09-22:
@dpgeorge could you have a look at this one?

dpgeorge, 2022-09-26:
Fixed by 769a90d6adf9e3fec5b3eadd6e153c674c3b57c6

closed 2022-09-26 by dpgeorge