microbit-foundation/micropython-microbit-v2 #135: see an download file text with editor V2

G-Boronad, 2022-09-27:
Hy, 
I'm physic teacher  and i use your editor with my students. I would like tu write text file (sensor data logging) on the card and then download the file.
Is there a way to do it with your editor (V2)?
Thanks for your job
G.Boronad

microbit-carlos, 2022-10-19:
Hi @G-Boronad,
You can use the data logging functionality to log sensor data and then visualise it or download it via the MY_DATA.HTM file in the MICROBIT drive:
- https://microbit.org/get-started/user-guide/data-logging/
- https://microbit-micropython.readthedocs.io/en/v2-docs/log.html

microbit-carlos, 2022-11-08:
I'll close this as answered, but please feel free to reopen the issue if your question wasn't really answered, or you have other questions.

closed 2022-11-08 by microbit-carlos