microbit-foundation/micropython-microbit-v2 #138: `ord()` difference between V1 and V2

microbit-carlos, 2022-10-24:
Originally reported in https://github.com/bbcmicrobit/micropython/issues/775 and related to https://github.com/bbcmicrobit/micropython/issues/774, but makes more sense in the V2 repo.

```
print("code point: {}".format(hex(ord("°"))))
```

In V1 we get the same as running the code in CPython:
```
code point: 0xb0
```

In V2:
```
TypeError: ord() expected a character, but string of length 2 found
```

microbit-carlos, 2022-10-24:
Maybe setting the `MICROPY_PY_BUILTINS_STR_UNICODE` flag in the V2 build could help with this issue.

dpgeorge, 2022-10-24:
Yes, setting that flag fixes this.  That should be set to match micro:bit v1.

Done in 60a991fd024e189ca8f199c45105df64d3c498a2

dpgeorge, 2022-10-24:
I also enabled `frozenset` which was missing compared to v1.  See 4e9c55024d78fa946d81ada060387f54d1a4c48a

closed 2022-10-24 by dpgeorge

microbit-carlos, 2022-10-25:
Thanks Damien!
Assuming this enables `frozenset`, @microbit-matt-hillsdon we might need to add this to the micro:bit stubs.