microbit-foundation/micropython-microbit-v2 #139: micro:bit freezes when playing `music.play([None])`

microbit-carlos, 2022-10-24:
Duplicating the issue here as it affect V2 as well:
- https://github.com/bbcmicrobit/micropython/issues/776

> The program
> 
> ```
> from microbit import *
> import music
> 
> music.play([None])
> ```
> 
> hangs the microbit

dpgeorge, 2022-10-24:
Using `None` in a tune was intended to be a "rest", ie no sound for a period.  But this was never documented and instead "r" and "R" are supported and documented characters for inserting a "rest".

Using `None` as a rest is actually broken, as seen by the above example (you can still use ctrl-C to stop it, but the rest goes for a very long time).

I suggest we just completely remove support for `None` in tunes.  It never worked and was never documented.  That will then fix this issue.

microbit-carlos, 2022-10-25:
Thanks Damien, sounds good 👍 
Just for curiosity, do we know that it was always broken, or has it been broken from a specific point in time?

dpgeorge, 2022-10-28:
> Just for curiosity, do we know that it was always broken, or has it been broken from a specific point in time?

As far as I can tell, this "rest" feature was broken since commit 0189136af557bc2c8514935f7351bcaafe32c47f done in Nov 2015, which first allowed music to be played in the background.  So I think it's safe to assume this feature was never used.

dpgeorge, 2022-10-28:
Fixed by d2130d82341b371a1721975ef0e0c3bd66d04231 (support for None as a rest is removed).

closed 2022-11-07 by dpgeorge