microbit-foundation/micropython-microbit-v2 #144: radio packets are being dropped

rhubarbdog, 2022-12-20:
Using 3 microbits, 2 transmitting and one counting packets, not all packets are being recieved by the counter.

y `yap.py` 19682 received 318 dropped
m `message.py` 1822 received 178 dropped

[counter.py.txt](https://github.com/microbit-foundation/micropython-microbit-v2/files/10268372/counter.py.txt)
[message.py.txt](https://github.com/microbit-foundation/micropython-microbit-v2/files/10268373/message.py.txt)
[yap.py.txt](https://github.com/microbit-foundation/micropython-microbit-v2/files/10268374/yap.py.txt)

rhubarbdog, 2022-12-20:
it's not just 2 microbit's communicating at the same time. I have this repo https://github.com/rhubarbdog/microbit-hands-up
i have 2 `pupil.py` micobit's and one `teacher.py`. i press the button on the pupil microbits a second appart.
When i send out the results from the `teacher.py` microbit the happy face isn't always shown

rhubarbdog, 2022-12-20:
also in the initial example `counter.py` counts message that are neither "yap." or "message." and receives none neither does `radio.receive()` throw an exception that a corrupt packet has been detected on the airwaves

martinwork, 2022-12-20:
microbit-hands-up fails because the length of str(id) is 35 but the message received is only 29 long.
`radio.config(length=40)` makes it work.

rhubarbdog, 2022-12-20:
On the microbit's i have the string `str(machine.unique_id())` must be less than 32 long because i have had both success and failure with 2 microbit's running `pupil.py` and one running `teacher.py`.

microbit-carlos, 2023-01-03:
Thanks for the additional info @rhubarbdog.
Would you be able to provide a minimum example where not more than one micro:bit is transmitting at the same time and a receiver still loses packets?

rhubarbdog, 2024-01-17:
transmitting with program ... on one microbit
```
from microbit import *
import radio

radio.config(data_rate = radio.RATE_2MBIT)
radio.on()

while True:
    if button_a.is_pressed():
        for _ in range(100):
            radio.send('message')
        while button_a.is_pressed():
            pass

    sleep(1)
```

and receiving on another microbit with program
```
from microbit import *
import radio

radio.config(length = 10, queue = 100, data_rate = radio.RATE_2MBIT)
radio.on()

count = 0 
while True:
    receive = radio.receive()
    if receive is None:
        sleep(1)
    else:
        count = count + 1

    if button_a.is_pressed():
        display.scroll(str(count))
```

before i added the `radio.config` argument `queue = 100` packets were being dropped at `data_rate` 1 MBit when i change it to 2 Mbit most packets sometimes all are being dropped

martinwork, 2024-01-17:
Should "length = 10" be included in both configs?