microbit-foundation/micropython-microbit-v2 #148: music defaults to pin0

rhubarbdog, 2022-12-22:
```
from microbit import *
import music

music.play('["c"], pin2, False)
```

plays music through the speaker on pin0

rhubarbdog, 2022-12-22:
i have this device [Kitronik GameZip 64  controler](https://kitronik.co.uk/products/5626-game-zip-64-for-the-bbc-microbit)
neopiexls are on pin0
external speaker pin2

this program seems to work
[gamezip.py.txt](https://github.com/microbit-foundation/micropython-microbit-v2/files/10285709/gamezip.py.txt)

but in bigger programs neopixels fail to light and the speaker latches on with a quiet buzzing noise

DaveAtKitronik, 2022-12-22:
Do you have an example that fails?

rhubarbdog, 2022-12-22:
Yes repository https://github.com/rhubarbdog/microbit-two-weeks it's my fortnight clone client server game.

Put `host.py` on one microbit and `gamezip.py` on the other. You'll need to edit `gamezip.py` and uncomment all lines with the word 'music'

DaveAtKitronik, 2022-12-23:
the linked code doesn't seem to work either way - I get a set of 3 leds on the Ubit display, but nothing else. the 'Host set is rotating, and the other set is static.
Can you make a simple this works, this doesn't work pair for a test case?

microbit-carlos, 2023-01-03:
Hi @rhubarbdog, if you could provide a simple programme to replicate the issue it would help identify the problem.

As far as `music.play()` or `audio.play()` always outputting sound via pin0 I haven't been able to replicate this. I've tried it with a micro:bit V2 and using a Kitronik KLEF Piano, which connects the external speaker to pin0.

Running this code as a programme or in the REPL and I don't hear sound from the external speaker. 

```
from microbit import *
import music

music.play(["c"], pin2, False)
audio.play(Sound.HAPPY, pin=pin2)
```

Did you test this with a micro:bit V2? Are you sure the sound you are hearing is not coming from the micro:bit on-board speaker? You can turn off the on-board speaker with `speaker.off()` to check.

rhubarbdog, 2024-01-22:
this has been solved with firmware version 2.1.2

closed 2024-01-22 by rhubarbdog