microbit-foundation/micropython-microbit-v2 #149: lack of was_touched for pin_logo makes it less useful as a 'third button'

jaustin, 2022-12-29:
For the buttons we can easily write code that's not super time-dependent wrt things happening in the main loop, by using `was_pressed`. However, with the pin_logo, as far as I can see can only test whether it `is_touched`.

I wouldn't be against having a was_pressed for all touch pins, either. Does anyone recall the reason for not having `was_` methods for pins?

microbit-carlos, 2023-01-03:
I don't recall there being a specific reason why `pin_logo.was_pressed()` was not added, I think it was mostly a matter of not being specified.

The rational at the time was to implement `pin_logo` similarly to the pin0/1/2,  as a `MicroBitTouchPin` instance, as that is really what it is. The only class with a `was_pressed()` method is the `MicroBitButton`, so it is not present in the large pins either:
- https://microbit-micropython.readthedocs.io/en/v2-docs/pin.html#microbit.MicroBitTouchPin
- https://microbit-micropython.readthedocs.io/en/v2-docs/button.html#Button.was_pressed

So if we'd like to implement this it would have to be by adding a `was_touched` method to `MicroBitTouchPin`, which would apply to pin 0/1/2 as well.

jaustin, 2023-01-04:
I'm fully supportive of adding was_touched to all pins, including the logo, unless it presents significant size or performance issues on V1/V2.

dpgeorge, 2023-01-16:
For V1 this should be straightforward, because all the code is already there to debounce touch and compute the "was touched" state.  It just has to be wired up for P0/P1/P2.

For V2 it depends on the CODAL.  The logo should work fine because `uBit.logo` is a `TouchButton` instance with a `wasPressed()` method.  For P0/P1/P2 it's not clear that it'll work because I don't see any instance of a `Button`/`TouchButton` for these pins.

microbit-carlos, 2023-01-19:
CODAL ticket:
- https://github.com/lancaster-university/codal-microbit-v2/issues/251

dpgeorge, 2023-02-13:
I assume we also want to implement `get_touches()` which will return the number of touches since the last time it was called.

microbit-carlos, 2023-02-13:
Yes, for symmetry we should add `get_touches()`.
Is `MicroBitButton.get_presses()` method implemented in MicroPython? So I assume adding it to the touch button doesn't require an additional CODAL method other than `wasTouched()`?

dpgeorge, 2023-03-13:
`was_touched()` and `get_touches` have now been implemented on v1.

dpgeorge, 2023-11-13:
CODAL as at v0.2.63 has added the `wasTouched()` method.

dpgeorge, 2023-11-13:
`was_touched()` and `get_touches()` have been implemented on `pin0`, `pin1`, `pin2` and `pin_logo` in 957e810818f239fddf8115014bb9b650ccf278cb

For `pin_logo` the code now uses `uBit.io.logo.isTouched()` rather than `uBit.logo.buttonActive()` to implement the Python-level `pin_logo.is_touched()` method.  I'm not sure if this makes a difference?  From my testing the new version seems less sensitive.

microbit-carlos, 2024-03-06:
The only thing left to close this task is:
- https://github.com/lancaster-university/codal-microbit-v2/issues/411

microbit-carlos, 2024-08-08:
> The only thing left to close this task is implementing this in CODAL and revert [3074c81](https://github.com/microbit-foundation/micropython-microbit-v2/commit/3074c81e385cec0db0c493f093263f9c34e35912):

Done in https://github.com/microbit-foundation/micropython-microbit-v2/pull/213.

closed 2024-08-08 by microbit-carlos

kevinjwalters, 2025-04-11:
Is there user feedback on this? I wanted to implement a touch and measure how long the logo was pressed for and did this with `pin_logo.is_touched()`. Unless presses were very quick with finger cleanly withdrawn I got a lot multiple presses.