microbit-foundation/micropython-microbit-v2 #150: `display.clear()` does not stop `display.show(loop=True, wait=False)`

microbit-carlos, 2023-03-17:
Reported in the V1 repository, but I think it's actually a V2 only issue:
- https://github.com/bbcmicrobit/micropython/issues/789

With this programme pressing button A should stop the animation looping using `display.clear()`, but it only clear the current image displayed, the next animation image is loaded.
Pressing B replaces the animation by just displaying a letter and at that point `display.clear()` works as expected.
```python
from microbit import *

display.show("hello", loop=True, wait=False)
while True:
    sleep(100)
    if button_a.is_pressed():
        display.clear()
    if button_b.is_pressed():
        display.show("X")
```

In V1 `display.clear()` does stop the animation.

davidtkaye, 2023-03-18:
Can confirm this is a problem on my V2 microbit, also coincidentally shows up on the microbit simulator in the online Python editor (where I first identified it).

dpgeorge, 2023-03-27:
Thanks for the report.  Should be fixed by a5dd649283ad6c232e5f924af3b24242f58e3bbc

dpgeorge, 2023-03-27:
@microbit-matt-hillsdon for the simulator you can now remove `microbit_hal_display_clear()`, per 5177bdc96df45266f344b2813f2f4d94891c7928

closed 2023-03-27 by dpgeorge