microbit-foundation/micropython-microbit-v2 #151: Clap detection

microbit-carlos, 2023-03-24:
CODAL has added a "clap" event, similar to "loud" and "quiet". 

https://github.com/lancaster-university/codal-core/commit/99b9399857fd5e6110c2f0b9f2b3d134cd59e92d

microbit-carlos, 2024-04-30:
For the `set_threshold()` method we'll need to throw an exception if the "clap" event is provided, as that threshold cannot be configured.

dpgeorge, 2024-05-20:
`SoundEvent.CLAP` is added in 8b8c1d3802164abe40659edde9cf3a717aabfde4

> For the `set_threshold()` method we'll need to throw an exception if the "clap" event is provided,

This is also done (didn't need any changes, was already raising an exception for invalid sound events).

dpgeorge, 2024-09-09:
@microbit-carlos this is already done (see above).  I tested it and it seems to work, `microphone.get_events()` does emit a `SoundEvent.CLAP` on occasion.