microbit-foundation/micropython-microbit-v2 #154: mpy-cross: error: variable 'xxx' set but not used [-Werror,-Wunused-but-set-variable

microbit-carlos, 2023-04-25:
```
../py/objgenerator.c:102:66: error: variable 'n_kwonly_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                                 ^
../py/objgenerator.c:102:54: error: variable 'n_pos_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                     ^
../py/objgenerator.c:102:21: error: variable 'n_exc_stack_unused' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                    ^
../py/objgenerator.c:102:81: error: variable 'n_def_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                                                ^
../py/objgenerator.c:102:41: error: variable 'scope_flags' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                        ^
5 errors generated.
make: *** [build/py/objgenerator.o] Error 1
```

Haven't had a chance to look at this properly, but I've updated the Xcode toolchain a few weeks ago and I'm guessing a new version of clang might be throwing these error/warnings now.

dpgeorge, 2023-05-08:
We should update the MicroPython submodule that this repository uses.  MicroPython v1.20.0 was released a few weeks ago and I suggest using this version.

Such an update will bring in bug fixes, more Python compatibility (eg `bytes.fromhex` and friends) and optimisations.  But it shouldn't break any existing micro:bit scripts.

microbit-carlos, 2023-05-08:
Sounds good 👍

dpgeorge, 2023-05-16:
Fixed by 16fe526d88175ab79a21cba8cc449ffea190f8d3 and 5176ffd761ecdcce9836f9a7e1f835710fb9e85c

microbit-carlos, 2023-07-10:
Thanks Damien! I can confirm I don't have mpy-cross compilation errors with the latest commit.

closed 2023-07-10 by microbit-carlos