microbit-foundation/micropython-microbit-v2 #155: `SoundEffect` fx value doesn't have an effect

microbit-carlos, 2023-05-04:
No matter what the `fx` value is, it doesn't change the sound.

```python
from microbit import *

fx_test = audio.SoundEffect(
    freq_start=3041,
    freq_end=3923,
    vol_start=59,
    vol_end=255,
    duration=500,
    waveform=audio.SoundEffect.WAVEFORM_SINE,
    fx=audio.SoundEffect.FX_NONE,
    shape=audio.SoundEffect.SHAPE_LINEAR,
)
audio.play(fx_test)
sleep(100)

fx_test.fx = audio.SoundEffect.FX_TREMOLO
audio.play(fx_test)
sleep(100)

fx_test.fx = audio.SoundEffect.FX_WARBLE
audio.play(fx_test)
sleep(100)

fx_test.fx = audio.SoundEffect.FX_VIBRATO
audio.play(fx_test)
```

[fx_test.hex.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/11399270/fx_test.hex.zip)

dpgeorge, 2023-05-08:
I had a play around with this.  The `fx` parameter is not doing anything because the "fx steps" and "fx param" parameters default to bad values.  Their defaults are 24 and 1 respectively, and they cannot be changed (from Python).

Do we need to expose these "fx steps" and "fx param" parameters to Python?  Or maybe give better defaults? (Or both).

microbit-carlos, 2023-05-15:
We can use the default values used by MakeCode, but if these are going to be the "unofficial" defaults and are not exposing them to the user, we should probably include them in CODAL instead.

https://github.com/microsoft/pxt/blob/5e2ff19c7d44bc659b997530c96b7ac9ad6a6551/webapp/src/components/soundEffectEditor/soundUtil.ts#L24-L42

martinwork, 2024-01-02:
Reported in support ticket https://support.microbit.org/helpdesk/tickets/70448 (private)

microbit-carlos, 2024-03-11:
@dpgeorge because these settings are different depending on the type of `fx` applied (tremolo, warble or vibrato):
https://github.com/microsoft/pxt/blob/5e2ff19c7d44bc659b997530c96b7ac9ad6a6551/webapp/src/components/soundEffectEditor/soundUtil.ts#L24-L42

CODAL doesn't have a way to set the same values as MakeCode without changing something in the CODAL API. We could either update CODAL to have something like an "effect factory", or maybe we could set up these fx-steps and fx-param values as macros defined in CODAL?
Do you have any thoughts, preferences or ideas?

dpgeorge, 2024-03-14:
> maybe we could set up these fx-steps and fx-param values as macros defined in CODAL?

Having simply a set of `#define`'s for the default fx-steps/fx-param values for each of tremolo, warble, vibrato, is a good solution.  That's very simple.

Then I guess we will not expose these fx-steps/fx-param values to the (Python) user, at least for now.  They will be set automatically when the fx is set.

dpgeorge, 2024-03-25:
In 8d9067d91bcfe316cd3aea5af1a873a37ca23d13 I made it so that when the `fx` is changed the param and steps values are updated to use default values corresponding to the given `fx` choice.  These default values follow what's used by MakeCode.

microbit-carlos, 2024-03-26:
CODAL PR:
- https://github.com/lancaster-university/codal-microbit-v2/pull/417

dpgeorge, 2024-08-26:
The above PR is merged and I think this issue can now be closed.

closed 2024-08-30 by microbit-carlos