microbit-foundation/micropython-microbit-v2 #156: CODAL heap used for audio recording

microbit-carlos, 2023-05-19:
CODAL allocates audio buffers in chunks in the heap during audio recording from the microphone.
MicroPython reserves the Python code heap with a 64 KB static array, and configures the MicroPython/CODAL stack to 8 KB.

If I remember correctly, the rest is left for CODAL heap, is that right?
I assume the MicroPython core itself (not the codal app port part) doesn't use the CODAL heap, is that correct?
In which case, that will mostly be used by CODAL code.

With all that in mind that might leave us around 32 KBs of heap (complete guesstimation, we should measure this), which is not a lot for recording audio.

It's also a bit weird if audio recording doesn't consume "user" memory, as reported by the micropython `machine` and `gc` modules. 

@dpgeorge @finneyj @JohnVidler what are our options here?

Will we need to somehow allocate recording buffers in the user's Python heap?
Is that possible with the current codebase? Will we need to hook custom allocators somewhere?

dpgeorge, 2023-06-12:
All the assumptions above are correct.  MicroPython has a fixed 64kB heap, and does not use the CODAL heap for anything except what CODAL uses "behind the scenes" when its API functions/methods are called.

For the audio buffers: allocating and freeing lots of large-ish sized buffers on the MicroPython heap could lead to heap fragmentation and out-of-memory issues, even if there is still enough free memory (due to fragmentation).  The way to prevent this is to either (1) have pre-allocated memory for the audio buffers, or (2) have a separate heap which is deterministic and used only for audio buffers.  The second option there is basically the same as using the CODAL heap.

How much memory is needed for audio buffers?  How much does it allocate each time (ie what are the sizes of the allocations)?

dpgeorge, 2024-03-25:
I think this issue can be closed.  The `audio-recording` branch is using MicroPython's GC heap to allocate memory for recording data.  The CODAL does not use much, just some temporary internal buffers.  This was made even better with `pullInto()`.

closed 2024-03-26 by microbit-carlos