microbit-foundation/micropython-microbit-v2 #158: set_pull(NO_PULL) behaves differently between V1 and V2

martinwork, 2023-06-14:
Arising from https://support.microbit.org/helpdesk/tickets/64103 (private)

I used the program below in a micro:bit with P0 connected to P1.
P1 sends it's analogue value to uart.
Press A to set P0 to NO_PULL.

In V1, P1's value drops as soon as A is pressed. 
In V2, P1's value remains steady until B is pressed to make P0 a digital input.

In MakeCode and C++, V1 behaves like V2
https://makecode.microbit.org/_g2c94FA4k6bE
[nopull.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/11749243/nopull.zip)

The program in the ticket was relying on the V1 behaviour, so stopped working with V2.

```
from microbit import *

pin0.write_digital(1)

while True:
    if button_a.was_pressed():
        pin0.set_pull(pin0.NO_PULL)
    if button_b.was_pressed():
        pin0.read_digital()
        
    uart.write("p1:"+str(pin1.read_analog())+"\r\n")

```

microbit-carlos, 2023-08-21:
So pin0 is set as an output high, and when then when the pull is changed to "no pull":
- V1 the value drops
- V2 the value stays high

Sounds like the bug is in MicroPython for V1 ? If the pin is output high, it should stay like that until the pin mod is changed, no?

> In MakeCode and C++, V1 behaves like V2
> https://makecode.microbit.org/_g2c94FA4k6bE
> [nopull.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/11749243/nopull.zip)

So MakeCode for V1 and V2 behave like MicroPython for V2, where the value stays high?

martinwork, 2023-08-21:
Yes, I think V1 has always had slightly odd behaviour because setting the pull also sets the pin to input, but the ticket report is that old code that has always worked with V1 doesn't work with V2.

If the V1 python behaviour was intentional, maybe V2 should be changed to match V1. If not, fixing V1 will mean the old code stops working with V1 too!

Rather than suggest a V1 bug fix, I posted here because it seemed the most likely place to look for a bridging solution: **to make code that behaves the same on V1 and V2, add read_digital() after set_pull()**.

Oh, wait! I didn't think to look before... The MicroPython documentation accessed from the Python Editor seems to say set_pull works in the V1 way: **"Calling set_pull will configure the pin to be in read_digital mode with the given pull mode."**

https://microbit-micropython.readthedocs.io/en/v2-docs/pin.html?highlight=set_pull#microbit.MicroBitDigitalPin.set_pull:~:text=Calling,given%20pull%20mode.