microbit-foundation/micropython-microbit-v2 #159: `microbit.set_volume()` halfs the value sent to `uBit.audio.setVolume()`

microbit-carlos, 2023-07-26:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/eeb1db318df7d678dbc7063934efe4258eaa70ab/src/codal_app/microbithal_audio.cpp#L78-L84

And `uBit.audio.setVolume()` input range is 0 to 255:
[codal-microbit-v2/source/MicroBitAudio.cpp#L161-L169](https://github.com/lancaster-university/codal-microbit-v2/blob/3fa546f7ef5fd34cb9167b41045f8d7b6235b67e/source/MicroBitAudio.cpp#L161-L169)
```cpp
int MicroBitAudio::setVolume(int volume)
{
    if (volume < 0 || volume > 255)
        return DEVICE_INVALID_PARAMETER;

    mixer.setVolume(volume*4);

    return DEVICE_OK;
}
```
And `Synthesizer::setVolume()` does take a value between 0 and 1023: [codal-core/source/streams/Synthesizer.cpp#L277-L289](https://github.com/lancaster-university/codal-core/blob/2658178aade744af248b116c2e4277b94a242b69/source/streams/Synthesizer.cpp#L277-L289)

So in a programme like this, we can hear the volume drop:

```python
from microbit import *
import music

music.pitch(440, duration=1000)
set_volume(255)
music.pitch(440, duration=1000)
```

dpgeorge, 2023-11-13:
Should be fixed by a37bc963d23dc2fb39fd7998775db003b52bdf12

dpgeorge, 2024-03-25:
This just needs testing @microbit-carlos

microbit-carlos, 2024-05-03:
Can confirm this is now working, thanks Damien!

closed 2024-05-03 by microbit-carlos