microbit-foundation/micropython-microbit-v2 #161: neopixel/ws2812 timings

microbit-carlos, 2023-10-30:
There is a wide range of ws2812 LEDs with slightly different timings and the CODAL version in the MicroPython v.2.1.1 release doesn't work with all of them:
- https://github.com/lancaster-university/codal-microbit-v2/issues/294

This has been fixed in the CODAL's PR https://github.com/lancaster-university/codal-nrf52/pull/47, but to do a low-risk v2.1.2 release we can apply the patch directly to v2.1.1 as done in: https://github.com/microbit-foundation/micropython-microbit-v2/compare/v2.1.1...0697c6d5b035e9369d3f2142b1d7a4cbe2301b11

microbit-carlos, 2023-10-30:
Okay, so I could not successfully trigger a new docker build of the MicroPython builds we have in https://github.com/microbit-foundation/micropython-build-env, because some pinned dependencies don't seem to be available in the ubuntu package repositories.
I'll need to look into that issue and fix it in a future date, but in the meantime I've created a v2.1.2 build by using the docker image created as part of the v2.1.1 build.

I was hoping to be able to checkout the [v2.1.x-branch](https://github.com/microbit-foundation/micropython-microbit-v2/compare/v2.1.x-branch), and re-build with only those changes, but for some reason that didn't triggering re-compilation for the codal-nrf52 files and ended up with a build with the only change being the version string.

Nevertheless, doing a clean build worked correctly, and there wasn't significant changes in the build output (sometimes, even with small changes, the compiler applies optimisation levels differently and/or moves things around, so so a binary/hex comparison is not easy).

The final hex output only contains 4 changes, so that made it really easy to verify everything else should be exactly the same. Which is important as I was hoping to be able to avoid having to run a lot of manual tests.

The v2.1.2 hex file can be downloaded from the GitHub release https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.1.2 and will be added to the Python Editor soon.

**The changes in the hex file are**:
- A change at address 0x29900:
  [image]
    - From the map file, is for the `neopixel_send_buffer` function, makes sense ✅.
      ```
      .text._Z20neopixel_send_bufferRN5codal3PinEPKhi
                      0x0000000000029880       0x84 /home/micropython-microbit-v2/lib/codal/build/libcodal-nrf52.a(neopixel.cpp.o)
                      0x0000000000029880                neopixel_send_buffer(codal::Pin&, unsigned char const*, int)
      .text._ZN5codal19allocate_peripheralEPv
                      0x0000000000029904       0x3c /home/micropython-microbit-v2/lib/codal/build/libcodal-nrf52.a(peripheral_alloc.cpp.o)
                      0x0000000000029904                codal::allocate_peripheral(void*)
      ```
- Another change at address 0x53A50, where we can clearly see the change from `WS2812B_LOW` and `WS2812B_HIGH` ([neopixel_ws2812b_timing.patch](https://github.com/microbit-foundation/micropython-microbit-v2/blob/0697c6d5b035e9369d3f2142b1d7a4cbe2301b11/src/neopixel_ws2812b_timing.patch)):
  [image]
    - This is the `codal::WS2812B::pull()` method, also expected ✅.
      ```
      .text._ZN5codal7WS2812B4pullEv
                      0x0000000000053a24       0x9c /home/micropython-microbit-v2/lib/codal/build/libcodal-nrf52.a(WS2812B.cpp.o)
                      0x0000000000053a24                codal::WS2812B::pull()
      .text._ZN5codal7WS2812B5_playEPKvib
                      0x0000000000053ac0       0x30 /home/micropython-microbit-v2/lib/codal/build/libcodal-nrf52.a(WS2812B.cpp.o)
                      0x0000000000053ac0                codal::WS2812B::_play(void const*, int, bool)
      ```
- The other two changes are strings changed as a result of updating the version number (https://github.com/microbit-foundation/micropython-microbit-v2/commit/0697c6d5b035e9369d3f2142b1d7a4cbe2301b11) ✅:
    - A change in the string `micro:bit v2.1.2+0697c6d on 2023-10-30; MicroPython v1.18 on 2023-10-30`:
      [image]
    - And a change in the string `MicroPython v1.18 on 2023-10-30; micro:bit v2.1.2 with nRF52833`:
      [image]

microbit-carlos, 2023-11-02:
@dpgeorge @jimmo v2.1.2 of MicroPython for micro:bit V2 has been deployed to https://python.microbit.org

closed 2023-11-02 by microbit-carlos

jimmo, 2023-11-03:
Thanks @microbit-carlos!! :)