microbit-foundation/micropython-microbit-v2 #162: micro:bit v2 MICROPYRHON onewire DB18B20 waterproof temp sensor

burksfamly, 2023-11-11:
I need micro:bit v2 micropython .py library files for the onewire protocol and DS18B20 driver for a waterproof temp sensor. Research has been frustrating. Has anyone ported this yet.

I took a look at PyCom. Looks good if I could get enough info on micro:bit v2 micropython implementation. For example need to deal with removing from machine import Pin with from microbit import pin1...and then deal wirh the equivalwnt OPEN_DRAIN pin mode using pin1...without a map of rhe micro:bit v2 micropython environment...this will be trial and error, and I am alreadg in too many rabbit holes on this project.  I DO NOT want to go back to the TMP36 because it requires a constant reference voltage. If you use your battery or battery points, temp readings change as the battery drains, and adding constant refererence voltage circuit is battery expensive

Any sources for the afforementioned files?

Thank You

martinwork, 2023-11-11:
Thanks to @rhubarbdog:
https://github.com/rhubarbdog/microbit-ds18b20

rhubarbdog, 2023-11-11:
The code doesn't work. It requires an oscilloscope to get the timings correct. The big number in the sleep sub routine needs to be reduced and possibly also the number of register 7 needs to be reduced before calling the sleep sub routine

martinwork, 2023-11-11:
Thanks for letting us know @rhubarbdog

dpgeorge, 2023-11-13:
MicroPython does have a built-in `onewire` module and drivers for the `ds18x20`.  But the `onewire` module is not enabled on the micro:bit and needs to be compiled into the firmware to work.

burksfamly, 2023-11-13:
Damien,
Thank you. Why was it 'turned off'? What would be required to get a micro:bit version of the micropython image onto the micro:bit. Will that cause problems with other services and hardware assumptions?
Maybe you could help with the actual dilemma I am having. I need a reliable temperature sensor for the micro:bit for HVAC supply and return temperatures as well as a waterproof implementation for aquariums. This sensor must be used on a battery powered micro:bit using micropython.
I had used the TMP36 sensor, but this requires a constant reference voltage, of which the micro:bit does not supply (without modification). So, when the TMP36 is used the temperature changes during battery drain based on the manner in which the analog reading is converted to temperature.
This is why I was exploring the use of the DS18B20 waterproof temperature probe which uses the onewire protocol and the DS18X20 driver unavailable in micropython for micro:bit.
So...I am in a catch 22. I either need a simple way to use the DS18B20 or a method to get a constant 3vdc reference voltage from the micro:bit to drive the TMP36 temperature sensor to get reasonably accurate temperature readings.
Any thoughts on your experience with this would be greatly appreciated.
I could use another micro:controller IF I could find one that is compatible with the micro:bit micropython radio communications protocol, of which I have not yet been able to find. In other words, I have not found another microcontroller configuration that will allow me to communicate via radio with the micro:bit v2 micropython environment.
Best,Randy

burksfamly, 2023-11-15:
https://support.microbit.org/helpdesk/tickets/68811     Hi Randall  Thanks for contacting the Micro:bit Educational Foundation.     Please don't CC GitHub comments to ***@***.*** Doing so creates new ticket that someone has to triage and assign.     All these people will see a notification of your comment automatically.  https://github.com/microbit-foundation/micropython-microbit-v2/watchers     Kind regards,  Martin Williams  Micro:bit Educational Foundation              
                           
                            On
                            Mon, 13 Nov at  2:06 PM

dpgeorge, 2023-12-18:
> Why was it 'turned off'?

Well, more so it was never turned on.  Micro:bit v1 didn't have it and neither does v2.

To enable it and get the DS18x20 driver working requires a bit of work:
- ability for pins to be in `OPEN_DRAIN` mode
- implement `mp_hal_pin_od_low`, `mp_hal_pin_od_high`
- implement `mp_hal_quite_timing_enter/exit`, `mp_hal_delay_us_fast`
- enable `MICROPY_PY_ONEWIRE`
- freeze in the `onewire.py` bus driver

Then, the `ds18x20.py` driver in `micropython-lib` will work.

leomlr, 2024-02-15:
Hi,

We look forward to being able to program the ds18b20 with micro:bit with either solution. While waiting for these updates, thank you Damien George and Phil Hal.

burksfamly, 2024-05-01:
Martin,
Good morning. I hope all is well.
Could you please explain to me how the micro:bit v2 memory is available and taken up in the micropython environment.I have researched this morning and cannot find a definitive answer:
micro:bit v2 using the micropython v3 environment
From sources:Total available Flash ROM: 512KBTotal available RAM: 128KB

but my confusion is what is earmarked for what and what is taken up already.
For example:
There is a "drive" available for sketch space (main.py, import *.py, user written files)How much of this is dedicated, available and from what block of memory type does it come from?
When micropython is loaded, how much space does it take up and from what type of memory?
There is a datalogging space that is earmarked. How large is this and from where does it come from?
So...for example...think of the following type of table
Flash ROM Available:                             512KB                            RAM Available:                                                                                128KB
Micropython Taks Up:                                -???                                -???The datalogger space taks up:                 -???                                 -???The "single folder drive" takes up             -???                                 -???Other items I haven't mention take up:    -???                                 -???
I am trying to determine how ALL types of memory is available and used my other than the user of the micro:bit v2.I am also trying to figure out        how much space is available for python files on the "single folder drive"
        how much space is available for run time memory include code, stack, data seg, and heap
        how much space is available for writing data (e.g., text and binary files to the "single folder drive, data to the datalogger)

I would also like to know if any of the available memory (e.g., data logger space) can be redefined for usefor example as additional storage for the "single folder drive" that could accommodate additional import *.py files.
Thank You,Randy
Message ID: ***@***.***>

martinwork, 2024-05-01:
Hi @burksfamly , you have sent your email to an unrelated issue on GitHub.
https://github.com/microbit-foundation/micropython-microbit-v2/issues/162
Would you like to send it to help@microbit.org? Then we might delete it from the issue.