microbit-foundation/micropython-microbit-v2 #167: i2c pins mode is always `i2c`

microbit-carlos, 2024-02-26:
With the update from https://github.com/microbit-foundation/micropython-microbit-v2/issues/112 we can change the I2C pins to other pins than 19 and 20. However, the pin modes are not updated.

Using a build from the latest commit in main:
https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/7565598370

```
>>> pin19.get_mode()
'i2c'
>>> pin0.get_mode()
'unused'
>>> i2c.init(sda=pin0, scl=pin1)
>>> i2c.scan()
[]
>>> pin0.get_mode()
'unused'
>>> pin0.write_digital(1)
>>> pin19.get_mode()
'i2c'
>>> pin19.write_digital(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin 19 in i2c mode
```

microbit-carlos, 2024-02-27:
@dpgeorge is it possible that this issue is caused by:
- https://github.com/lancaster-university/codal-microbit-v2/issues/219

dpgeorge, 2024-03-25:
The problem here is MicroPython specific, it's because previous I2C and SPI pins are not being released, and new ones not being acquired.

dpgeorge, 2024-03-25:
Fixed by af721c760d8663195e749c7e2b4d97d4f632c258

closed 2024-03-25 by dpgeorge