microbit-foundation/micropython-microbit-v2 #168: `uBit.io.logo` might need to be configured as capacitive mode on startup

microbit-carlos, 2024-02-26:
It's possible that with the move to `uBit.io.logo` the default NRF52Pin touch mode is resistive and it might need to be set up to capacitive on startup.

With the latest commit in main, running this code and touching the logo is only registered if I am touching the GND pin as well:

```python
from microbit import *
while True:
    print(pin_logo.is_touched())
    sleep(200)
```

microbit-carlos, 2024-03-06:
There might be some technical reason why Capacitive might or might not be the default for `uBit.io.logo`, which we'll need to discuss in:
- https://github.com/lancaster-university/codal-microbit-v2/issues/411

In the meantime I think we can probably change the mode in the MicroPython side, and if the default changes in the future we can revert it later.

dpgeorge, 2024-03-25:
Actually, the problem here is a bit subtle.  `uBit.io.logo` is already of type `NRF52Pin` and MicroPython is calling `NRF52Pin::wasTouched()` to get touch information for the `pin_logo.was_touched()` method.  That function's signature is:
```c
int wasTouched( TouchMode touchMode = TouchMode::Capacitative );
```
So we are already using capacitive touch for `pin_logo.was_touched()`.  Just not `pin_logo.is_touched()`.

If you first run `pin_logo.was_touched()` it will configure it to capacitive, and then `pin_logo.is_touched()` actually works in touch mode!

dpgeorge, 2024-03-25:
My above statement seems to be true for all pins, not just `pin_logo`.  Eg, run the above while-True loop using `pin0` and then run `pin0.was_touched()` and then run the loop again.  The second time the pin seems much more responsive.

dpgeorge, 2024-03-25:
In 3074c81e385cec0db0c493f093263f9c34e35912 I made `pin_logo` capacitive at start up.

The remaining thing to do is fix CODAL so that calling `wasTouched()` without any arguments doesn't change the pin mode.

microbit-carlos, 2024-03-26:
Thanks Damien, once we resolve https://github.com/lancaster-university/codal-microbit-v2/issues/411 in CODAL we can also revert that change.
`wasTouched()` is tracked in:
- https://github.com/lancaster-university/codal-microbit-v2/issues/416

microbit-carlos, 2024-08-08:
Workaround removed in https://github.com/microbit-foundation/micropython-microbit-v2/pull/213.

closed 2024-08-08 by microbit-carlos