microbit-foundation/micropython-microbit-v2 #169: When checking microphone SoundEvents there is no way to "stop listening"

microbit-carlos, 2024-03-05:
For example simply running `microphone.current_event()` turns on the microphone and there is no way to turn it off.

microbit-carlos, 2024-03-21:
@dpgeorge I was wondering if we should change `microphone.stop_recording()` to simply be `microphone.stop()` so that it could also apply for this use case where we might not want to listen for microphone events in the background?

dpgeorge, 2024-03-25:
We could add a separate `microphone.stop_listening()` method.  Then the microphone will have two modes that it can independently start/run/stop: listening and recording.

I'm not sure how useful it is to have them separated.  Is there ever a case where you always want to be listening for SoundEvents, even during recording, and when recording stops you want to keep listening?  Maybe.

Also, if you use `microphone.record_into(wait=True)` and it stops automatically, or you use `microphone.record()` (which is blocking), does it implicitly turn off listening when it finishes recording?

microbit-carlos, 2024-03-25:
Agreed, the user might want to control the microphone events and recording independently, so we should have two independent functions: `stop_recording()` and `stop_listening()`