microbit-foundation/micropython-microbit-v2 #170: Add option to get Sound Level in decibels (dB)

microbit-carlos, 2024-03-19:
CODAL has the option choose between 8-bit and dB for the LevelDetector. We had a couple of requests and it's already built in CODAL, so it'd be good to have it in the next beta.

API Options:
- Adding an argument to `microphone.sound_level()`, possibly `microphone.sound_level(db=True)`?
- Creating a new function like `microphone.sound_level_db()`
- Change units via different method, e.g. `microphone.set_sound_level_unit(microphone.SOUND_LEVEL_DB)`
    - This would be my least preferred option

dpgeorge, 2024-03-20:
My preference would be `microphone.sound_level_db()`.  It's simple.  And it makes it very clear what the units are.

microbit-carlos, 2024-03-21:
Agreed:
- https://github.com/bbcmicrobit/micropython/pull/803

dpgeorge, 2024-09-09:
See #221 for an implementation.

dpgeorge, 2024-09-16:
Implemented in e876dcac96d597fb35f7541f752c2ea34dcae854

closed 2024-09-16 by dpgeorge