microbit-foundation/micropython-microbit-v2 #173: Doesn't build on Windows

microbit-carlos, 2024-03-22:
Tries on two different set ups.

This one with Mysy2, CMake 3.28.1, Python 3.8 (installed via Windows installer)

```
Using library: codal-microbit-nrf5sdk
-- Configuring done (1.0s)
-- Generating done (1.0s)
-- Build files have been written to: /c/workspace/micropython-microbit-v2/src/build
git -C ../lib/codal checkout CMakeLists.txt
Updated 1 path from the index
git -C ../lib/codal checkout CMakeLists.txt
Updated 0 paths from the index
make -C codal_port
make[1]: Entering directory '/c/workspace/micropython-microbit-v2/src/codal_port'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
(cd ../../lib/micropython && py py/makeversionhdr.py /c/workspace/micropython-microbit-v2/src/codal_port/build/genhdr/mpversion.h)
py make_microbit_version_hdr.py build/genhdr/microbitversion.h
GEN build/genhdr/qstr.i.last
Traceback (most recent call last):
  File "../../lib/micropython/py/makeqstrdefs.py", line 213, in <module>
    preprocess()
  File "../../lib/micropython/py/makeqstrdefs.py", line 83, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "C:\Users\IEUser\AppData\Local\Programs\Python\Python38\lib\multiprocessing\pool.py", line 86
8, in next
    raise value
  File "C:\Users\IEUser\AppData\Local\Programs\Python\Python38\lib\multiprocessing\pool.py", line 12
5, in worker
    result = (True, func(*args, **kwds))
  File "../../lib/micropython/py/makeqstrdefs.py", line 65, in run
    return subprocess.check_output(args.pp + flags + files)
  File "C:\Users\IEUser\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 415, in chec
k_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\IEUser\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\IEUser\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 858, in __in
it__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\IEUser\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 1311, in _ex
ecute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
make[1]: *** [../../lib/micropython/py/mkrules.mk:122: build/genhdr/qstr.i.last] Error 1
make[1]: *** Deleting file 'build/genhdr/qstr.i.last'
make[1]: Leaving directory '/c/workspace/micropython-microbit-v2/src/codal_port'
make: *** [Makefile:44: libmicropython] Error 2
```

Cygwin with CMake 3.14.3, Python 3.8 (installed via Windows installer):
```
Using library: codal-microbit-nrf5sdk
-- Configuring done
-- Generating done
-- Build files have been written to: C:/workspace/micropython-microbit-v2/src/build
git -C ../lib/codal checkout CMakeLists.txt
Updated 1 path from the index
git -C ../lib/codal checkout CMakeLists.txt
Updated 0 paths from the index
make -C codal_port
make[1]: Entering directory '/cygdrive/c/workspace/micropython-microbit-v2/src/codal_port'
py ../../lib/micropython/py/makeversionhdr.py build/genhdr/mpversion.h
GEN build/genhdr/mpversion.h
(cd ../../lib/micropython && py py/makeversionhdr.py /cygdrive/c/workspace/micropython-microbit-v2/src/codal_port/build/genhdr/mpversion.h)
GEN /cygdrive/c/workspace/micropython-microbit-v2/src/codal_port/build/genhdr/mpversion.h
Traceback (most recent call last):
  File "py/makeversionhdr.py", line 148, in <module>
    main()
  File "py/makeversionhdr.py", line 144, in main
    make_version_header(args.repo_path, args.dest[0])
  File "py/makeversionhdr.py", line 127, in make_version_header
    with open(filename, "w") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/cygdrive/c/workspace/micropython-microbit-v2/src/codal_port/build/genhdr/mpversion.h'
make[1]: *** [Makefile:136: build/genhdr/microbitversion.h] Error 1
make[1]: Leaving directory '/cygdrive/c/workspace/micropython-microbit-v2/src/codal_port'
make: *** [Makefile:44: libmicropython] Error 2

```

On GitHub Actions:
https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/8391892458/job/22983253141
```
Using library: codal-microbit-nrf5sdk
-- Configuring done (38.1s)
-- Generating done (0.1s)
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_BUILD_TYPE

-- Build files have been written to: D:/a/micropython-microbit-v2/micropython-microbit-v2/src/build
git -C ../lib/codal checkout CMakeLists.txt
Updated 1 path from the index
git -C ../lib/codal checkout CMakeLists.txt
Updated 0 paths from the index
C:/mingw64/bin/make -C codal_port
Makefile:140: *** target pattern contains no '%'.  Stop.
make[1]: Entering directory 'D:/a/micropython-microbit-v2/micropython-microbit-v2/src/codal_port'
make[1]: Leaving directory 'D:/a/micropython-microbit-v2/micropython-microbit-v2/src/codal_port'
make: *** [Makefile:44: libmicropython] Error 2
```

microbit-carlos, 2024-03-22:
Okay, at least the Msys2 system does work when installing Python internally with Pacman instead of using the Windows system Python. So it's more an issue about with using system Python inside Msys2 (although maybe setting the subprocess `shell` argument to `True` would be worth trying).

dpgeorge, 2024-03-25:
For the first case:
```
FileNotFoundError: [WinError 2] The system cannot find the file specified
```
It's hard to tell exactly what it can't find, but it may be referring to the C preprocessor, named by the variable `$(CPP)` in the makefile.

In the second case (cygwin) it looks something to do with relative vs absolute paths and the destination build directory.