microbit-foundation/micropython-microbit-v2 #176: Can't build on macOS: fatal error: stdint.h: No such file or directory

telephono, 2024-03-25:
I'm unable to build `micropython-microbit-v2` on macOS 13 (x86_64) or macOS 14 (arm64) but it might be a user error.

It look's like it can't find the Standard Library?

There are hundreds of error messages like this:
```
[...]
In file included from ../../lib/micropython/py/mpstate.h:29,
                 from ../../lib/micropython/py/runtime.h:29,
                 from ../../lib/micropython/extmod/machine_adc.c:27:
/usr/local/Cellar/arm-none-eabi-gcc/13.2.0/lib/gcc/arm-none-eabi/13.2.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
[...]
```

This is what I have installed:
```shell
❯ xcode-select -p
/Library/Developer/CommandLineTools
```

```shell
❯ brew list | grep arm-none
arm-none-eabi-binutils
arm-none-eabi-gcc
arm-none-eabi-gdb
```

dpgeorge, 2024-03-25:
You'll need to install `newlib`, or some form of it.

Maybe this can help: https://stackoverflow.com/questions/77249021/no-stdint-h-in-arm-none-eabi-gccmac

telephono, 2024-03-26:
Thank you, this did the trick.

closed 2024-03-26 by telephono