microbit-foundation/micropython-microbit-v2 #178: Are all `time.tick_xxx()` functions bound to 30 bits?

microbit-carlos, 2024-04-03:
Looking at `time.tick_cpu()` and the values returned, it looks like this functions overflows at the 30-bit limit of a MicroPython "small int".
Would this be true for all `time.tick_xxx()`? If that's the case, it's something we could add to the docs, as they are currently vague.

microbit-carlos, 2024-04-03:
Ah, looking at the upstream code it's more clear that this is the case:
https://github.com/micropython/micropython/blob/v1.22.0/extmod/modtime.c#L151-L164
```c
STATIC mp_obj_t time_ticks_ms(void) {
    return MP_OBJ_NEW_SMALL_INT(mp_hal_ticks_ms() & (MICROPY_PY_TIME_TICKS_PERIOD - 1));
}
MP_DEFINE_CONST_FUN_OBJ_0(mp_time_ticks_ms_obj, time_ticks_ms);

STATIC mp_obj_t time_ticks_us(void) {
    return MP_OBJ_NEW_SMALL_INT(mp_hal_ticks_us() & (MICROPY_PY_TIME_TICKS_PERIOD - 1));
}
MP_DEFINE_CONST_FUN_OBJ_0(mp_time_ticks_us_obj, time_ticks_us);

STATIC mp_obj_t time_ticks_cpu(void) {
    return MP_OBJ_NEW_SMALL_INT(mp_hal_ticks_cpu() & (MICROPY_PY_TIME_TICKS_PERIOD - 1));
}
MP_DEFINE_CONST_FUN_OBJ_0(mp_time_ticks_cpu_obj, time_ticks_cpu);
```

And https://github.com/bbcmicrobit/micropython/blob/v1.1.1/source/extmod/utime_mphal.c#L67-L80.

We can update the micro:bit MicroPython docs for V1 and V2 to reflect this.

dpgeorge, 2024-04-08:
The wrap around behaviour is documented in the MicroPython docs: https://docs.micropython.org/en/latest/library/time.html#time.ticks_ms