microbit-foundation/micropython-microbit-v2 #180: `audio.stop()` doesn't work

microbit-carlos, 2024-04-08:
Tested with v2.1.2 and the lates recording&playback branch. 

`audio.stop` doesn't work in this example:

```python
from microbit import *

while True:
    if button_a.is_pressed():
        audio.play(Sound.HAPPY, wait=False)
        # We can see here that the image is shown as soon as the sound starts playing
        display.show(Image.HAPPY)
        # But the sound plays to completion
        audio.stop()
        sleep(200)
    sleep(100)
    display.clear()
```

Simpler example in REPL (the next prompt `>>>` also appears before the sound stops playing):

```
>>> audio.play(Sound.HAPPY, wait=False); audio.stop()
>>> audio.play(Sound.HAPPY, wait=False); sleep(10); audio.stop()
>>> 
```

Replacing the play call with `audio.play(audio.SoundEffect(duration=3000), wait=False)` doesn't work either.

dpgeorge, 2024-04-30:
This is not related to the audio-recording branch, it's a bug in the master branch.

Now fixed on master by c143b24acf26fdebc6eb965c625430521d4a0c21

microbit-carlos, 2024-04-30:
Sorry, I forgot to mention that `microphone.stop_recording()` didn't work either:

```python
>>> a = microphone.record(4000)
>>> audio.play(a, wait=False); sleep(1000); microphone.stop_recording()
>>> # Recording plays for 4 seconds
```

microbit-carlos, 2024-04-30:
Oh yes, you are right, brain fart from my side!

closed 2024-04-30 by microbit-carlos