microbit-foundation/micropython-microbit-v2 #188: AudioFrame doesn't accept negative indexes or slices

microbit-carlos, 2024-04-09:
Tested with v2.1.2 and the latest version (at the time of writing) of the recording & playback branch: https://github.com/microbit-foundation/micropython-microbit-v2/commit/0b06914c71c18533da90df85230ac198578669bf

We can use positive array indexes to access and modify the data inside an AudioFrame, but not a negative index to access.

```
>>> af = audio.AudioFrame()
>>> af[len(af) - 1] = 123
>>> af[-1]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: index out of bounds
>>> af[1:]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert slice to int
>>> af[1:2]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert slice to int
>>> 
```

If we wanted to enabled this for AudioFrame we should probably enable it for V1 as well. Is the performance/size penalty for this significant?

Technically we could use a `memoryview` for this, but it's not a well know feature, so even if we documented well and add it to the examples it might not be used as much.
We also have issue https://github.com/microbit-foundation/micropython-microbit-v2/issues/187 released to memoryview.

microbit-carlos, 2024-04-29:
This will be covered as well with this new proposal:
- https://github.com/microbit-foundation/micropython-microbit-v2/issues/205

microbit-carlos, 2024-07-21:
No need to implement this in AudioFrame, just need to make sure it is in the AudioTrack implementation.

closed 2024-07-21 by microbit-carlos

dpgeorge, 2024-08-05:
AudioTrack's should now be able to be indexed with negative numbers.