microbit-foundation/micropython-microbit-v2 #200: `AudioFrame` with a small rate can result in a size of zero

microbit-carlos, 2024-04-24:
```
>>> len(AudioFrame(1000, 1))
32
>>> len(AudioFrame(999, 1))
0
>>> len(AudioFrame(1, 1))
0
>>> 
```

microbit-carlos, 2024-04-29:
Great, thanks Damien!

microbit-carlos, 2024-04-29:
> I have changed the code to ensure that the size is always at least 32 bytes.

Using a small `duration` or `rate` can still create an `AudioFrame` with less than 32 bytes:

```python
>>> len(AudioFrame(1000, 1))
1
>>> len(AudioFrame(1, 1))
1
>>> 
```

dpgeorge, 2024-04-30:
> Using a small `duration` or `rate` can still create an `AudioFrame` with less than 32 bytes:

This is intended.

Now that the code supports arbitrary byte lengths (ie no longer a multiple of 32), it's possible to create frames all the way down to 1 byte in size.

microbit-carlos, 2024-04-30:
Sounds good, thanks Damien!

closed 2024-04-30 by microbit-carlos