microbit-foundation/micropython-microbit-v2 #214: Doubling the rate of an AudioRecording doesn't half the playback time

microbit-carlos, 2024-08-08:
Dobling the AudioRecording rate should half the time it takes to play, however that's not the case and the sound playback sounds a bit weird, not just speed up, but almost like there are gaps in the sound.

```python
>>> r = microphone.record(2000)
>>> t = running_time(); audio.play(r); running_time() - t
1977
>>> r.set_rate(r.get_rate() * 2)
>>> t = running_time(); audio.play(r); running_time() - t
1410
>>> 
```

We can ignore that the original time is a bit off, as that would be covered by https://github.com/microbit-foundation/micropython-microbit-v2/issues/182.

dpgeorge, 2024-08-21:
I can reproduce the problem.  The same issue is apparent in a simple recording at 22kHz:
```py
print("recording...")
t = running_time()
r = microphone.record(1000, rate=22000)
dt = running_time() - t
print("record time:", dt)

print("playing...")
t = running_time()
audio.play(r)
dt = running_time() - t
print("play time:", dt)
```

The output is:
```
recording...
record time: 1015
playing...
play time: 2006
```
So it takes double the time to play the sample than it does to record it.

microbit-carlos, 2024-08-21:
Discussed in a call and we can increase the playback buffer to fix this issue.

dpgeorge, 2024-09-09:
I have implemented two separate improvements for this issue:
1. Increase playback buffer from 32 bytes to 64 bytes.
2. Optimise the case of playing a big buffer of data, to feed the CODAL audio pipeline as soon as it needs data.  This can be done because we don't need to call into Python to get the next frame, it's already available as data in the `AudioRecording`/`AudioTrack` buffer.

The latter is necessary when the playback rate is large, eg 30000 samples per second.

microbit-carlos, 2024-09-16:
Thanks Damien! I can confirm this has been fixed 🎉