microbit-foundation/micropython-microbit-v2 #215: Current audio recording sounds a bit quiet

microbit-carlos, 2024-08-08:
In comparison the recording in MakeCode is a bit louder, the background noise is also increased, but it makes voice recordings a bit easier to understand.

I've created this hex file for comparison using a [test version of MakeCode ](https://makecode.microbit.org/app/f650a417b2495e02bc009c72ccd5578b96c91a9d-9b51dbe296#editor)with the latest CODAL tag v0.2.67: [makecode-test-record.hex.zip](https://github.com/user-attachments/files/16554164/makecode-test-record.hex.zip)

Button A records, button b plays recording (logo touch changes the playback speed between normal [n], fast [f] and slow [s]).
Code can be seen here: https://makecode.microbit.org/_6x6Hq9EC8i6R

The MakeCode pipeline configuration code can be seen here: https://github.com/microsoft/pxt-microbit/blob/1049fc5ac641da9df65b2be174cbcec0cdfca017/libs/audio-recording/recording.cpp#L44-L56

Maybe it's the creation of a mixer channel with the StreamRecording and the volume configuration below that?

dpgeorge, 2024-09-10:
Possibly try `channel->setVolume(75.0)`.

dpgeorge, 2024-09-11:
MakeCode uses a sample range of 1023, yet MicroPython uses 255, so the corresponding volume increase would be `channel->setVolume(18.7)`.  I tried that setting but it makes the background noise very loud, much louder than the MakeCode example above.

I'm not sure what MakeCode is doing that it can keep the noise down while still amplifying the volume of the recording.

Also, because MicroPython uses the same audio output channel for all audio sources, if the volume is increased for recordings then it's increased for everything (because you can't tell if you're playing a recording or some other sample, eg loaded from a file).  Sound expressions have their volume increased as well.

So we might need to come up with a better way for MicroPython to make the recordings louder.  Maybe the samples coming out of the microphone can be automatically amplified (instead of amplifying the playback)?

microbit-carlos, 2024-09-24:
> Also, because MicroPython uses the same audio output channel for all audio sources, if the volume is increased for recordings then it's increased for everything (because you can't tell if you're [...]

But that would only be the case if we set the mixer global volume (`uBit.audio.mixer.setVolume()`) right?
I thought creating a mixer channel and changing the volume in the channel itself (`raw_source.channel->setVolume()`) would only affect that channel, no?

Since there are some differences between the MakeCode & MicroPython pipeline configuration I think it's fine to play with the numbers until it sounds around the same volume-ish. Had a quick play around and setting ``raw_source.channel->setVolume()` to around 4.0 sounds close enough. Does that also affect other channels like the Sound Expressions?

dpgeorge, 2024-09-26:
> I thought creating a mixer channel and changing the volume in the channel itself (`raw_source.channel->setVolume()`) would only affect that channel, no?

You are right that sound expressions will not be affected.  Also speech is on a different channel.

So it's only the AudioFrame/AudioRecording/AudioTrack sound level that will be changed.

> Had a quick play around and setting ``raw_source.channel->setVolume()` to around 4.0 sounds close enough. Does that also affect other channels like the Sound Expressions?

No, that should not affect sound expressions, nor speech.