microbit-foundation/micropython-microbit-v2 #223: Playing an empty AudioFrame never returns

microbit-carlos, 2024-09-16:
This in the REPL never returns until Ctrl+C is sent:
```
>>> audio.play(audio.AudioFrame())
```

But as soon as something else larger is played (like Sound.HAPPY) it does work until the next power cycle.

Might be related to increasing `AUDIO_OUTPUT_BUFFER_SIZE` to 64, as `audio.play(audio.AudioTrack(bytearray(63)))` doesn't return either, but `audio.play(audio.AudioTrack(bytearray(64)))` does.

dpgeorge, 2024-09-18:
I can reproduce the problem.  It looks like a CODAL issue and related to the recently fixed #182 which used `uBit.audio.isPlaying()` to check if audio is playing or not.

It seems that upon startup, CODAL reports that audio is playing:
```c
int main() {
    uBit.init();
    uBit.audio.isPlaying(); // returns true!
    return 0;
}
```
I can work around this, but it should probably be fixed in CODAL.

dpgeorge, 2024-09-18:
> It seems that upon startup, CODAL reports that audio is playing

And this is reflected in MicroPython by doing `audio.is_playing()` after a reset, it returns `True`.

dpgeorge, 2024-09-18:
On the `audio-recording` branch, I have fixed the issue with short audio not returning, by fixing a related issue where the last part of audio may not have been sent out the pipeline if it was shorter than 64 bytes.

But, the issue with `audio.is_playing()` returning `True` after a reset still remains.