microbit-foundation/micropython-microbit-v2 #225: Slow time on micro:bit V2

kevinjwalters, 2025-02-20:
I've observed that time is slow _in micropython_ on the micro:bit V2.

This program using `sleep` runs for longer than 100 seconds.

```
from microbit import sleep

sleep(15 * 1000)
pause_ms = 100 * 1000

count = 1
while True:
    print("{:06d}".format(count))
    sleep(pause_ms)
    count += 1
```

It produces this output (borrowed the Arduino IDE as it has timestamping, running on Windows 11)

```
20:44:16.077 -> 000001
20:45:56.757 -> 000002
20:47:37.501 -> 000003
20:49:18.167 -> 000004
20:50:58.920 -> 000005
20:52:39.635 -> 000006
20:54:20.306 -> 000007
```

The 100 second sleep (plus a tiny bit of processing for format/print/increment) is 100.680, 100.744, 100.666,  100.753, 100.715, 100.671. The clock appears to be running about 7.0% slow (at 23C). It's easy to dismiss this as poor time keeping on a microcontroller but the micro:bit V2 has a crystal (listed in BOM) and all is fine in CircuitPython on same board.

I first noticed this when checking the performance of the MCP7940N rtc on a ZIP Halo HD vs MicroPython `utime.ticks_us()`. Across 24 hours the RTC is running about 18 seconds fast (I think this has a trim feature which I may explore to get it spot on), MicroPython is 559 seconds slow (about 6.5%) and CircuitPython (`time.monotonic_ns()`) is just 2 seconds slow.

[image]

@JackAtKitronik might be interested in this.

kevinjwalters, 2025-02-20:
I wasn't expecting this but just tried on a V1 and same thing.

```
21:17:05.310 -> 000001
21:18:45.989 -> 000002
21:20:26.738 -> 000003
21:22:07.441 -> 000004
21:23:48.167 -> 000005
21:25:28.859 -> 000006
21:27:09.597 -> 000007
````

That's 100.679, 100.749, 100.703, 100.726, 100.692, 100.738 seconds.

kevinjwalters, 2025-02-21:
For comparison, [MakeCode ](https://makecode.microbit.org/_Ykg4oweM03cf) on V2: 

```
13:39:20.760 -> 000001
13:41:00.739 -> 000002
13:42:40.758 -> 000003
13:44:20.778 -> 000004
13:46:00.817 -> 000005
13:47:40.861 -> 000006
13:49:20.870 -> 000007
```

That's 99.979, 100.019, 100.020, 100.039, 100.044, 100.009ms.

kevinjwalters, 2025-02-25:
My V2 is busy at the moment testing a trimmed MCP7940N. The same V1 is still slow with this code

```
from microbit import sleep
import radio

radio.config(group=23)
radio.on()

sleep(15 * 1000)
pause_ms = 100 * 1000

count = 1
while True:
    print("{:06d}".format(count))
    sleep(pause_ms)
    count += 1
````

produces

```
15:00:50.008 -> 000001
15:02:30.700 -> 000002
15:04:11.390 -> 000003
15:05:52.107 -> 000004
15:07:32.838 -> 000005
```

on `MicroPython v1.9.234-gd64154c73 on 2017-09-01; micro:bit v1.1.1 with nRF51822`.

kevinjwalters, 2025-02-25:
From my first tests on V2 I have a linear corrected version of graph and that does show MicroPython is all over the place which supports the explanation of a mediocre internal oscillator.

[image]

kevinjwalters, 2025-03-13:
I think this was discovered back in 2016 too on the V1 in [REUK.co.uk: Testing Accuracy of micro:bit Internal Clock](https://www.reuk.co.uk/wordpress/testing-accuracy-of-microbit-internal-clock/), that one was out by `(600/598.51 -1) * 1000 * 1000 = +2489.5ppm`

martinwork, 2025-04-14:
An alternative method to monitor micro:bit's time:  [millis.zip](https://github.com/user-attachments/files/19743598/millis.zip)

Install millis-python.hex in micro:bit V2. Run millis.py on PC. The PC periodically asks micro:bit for it's millis clock and compares it's elapsed time. Disconnect WebUSB or it will corrupt the python serial.

It's clear within a minute that the micro:bit V2 is not quite 1% out.

Edit millis-python.hex to #import radio, and test again. Radio enables the HFCLK in V2, and the micro:bit times seem to match the PC to the millisecond, though I haven't run the test for more than a few minutes,

I think micropython is using a version of CODAL before the HFCLK was always enabled:

https://github.com/lancaster-university/codal-microbit-v2/blob/d159aba47e1797ace83ffbac81009909326f3c4f/model/MicroBit.cpp#L291

https://github.com/lancaster-university/codal-microbit-v2/commit/ed14fa2c330a5c1b9a2904b922823a1112e20dfe

microbit-carlos, 2025-04-25:
Thanks Martin! I had forgotten newer CODAL releases had enabled the crystal on `uBit.init()`.

@kevinjwalters if you try any of the MicroPython builds from this CI job, it should use the crystal by default and provide more accurate timings (so the next MicroPython release will have this by default as well):
https://github.com/microbit-carlos/micropython-microbit-v2/actions/runs/14669538988

kevinjwalters, 2025-06-04:
@microbit-carlos I have tried a recent `v1.23.0` build which @martinwork made. When my code in Arduino IDE (for timestamping) it looked much improved. _BUT_ there's a curious anomaly between two of my micro:bit. By luck I had some code that times some maths and it takes a different about of time on this version between two micro:bits.

```
ZHH-B
R24 MATH 481399
 
ZHH-D
R24 MATH 455705
```

The full output is in https://github.com/lancaster-university/codal-microbit-v2/issues/475#issuecomment-2920470579

kevinjwalters, 2025-08-27:
I've did some clock testing off a recent build and it now looks good and like what you'd expect, six V2 boards here ranging from +8.88ppm to +16.60ppm, all within a typical +/- 20ppm for a crystal.

[image]