microbit-foundation/micropython-microbit-v2 #226: Increase file system size on V2

kevinjwalters, 2025-02-26:
The flash storage increased from 256 on the V1 to 512KB on the V2. The file system still appears to be limited to "there is approximately 30k of storage available" according to https://microbit-micropython.readthedocs.io/en/latest/tutorials/storage.html.

I ran out of space for wav files for a [previous project](https://www.instructables.com/PIR-Movement-Alarm-With-BBC-Microbit-and-External-/). I'm now running out of space for code with the double whammy of not being able to use mpy (https://github.com/bbcmicrobit/micropython/issues/524).

kevinjwalters, 2025-02-28:
For anyone else facing code size issues, I have found that `pyminify` from https://pypi.org/project/python-minifier/ helps.

microbit-carlos, 2025-04-01:
Thanks for the request @kevinjwalters!
Yeah, we should consider adding a compile time flag to expand the fs to all the flash left available. That way advance users can easily compile MicroPython and take advantage of the extra space.

That being said, in V2 we also include the Nordic SoftDevice and bootloader, so that adds an extra 150-ish KBs. Between that and the additional features the last time I checked the max fs size possible was around 60 Kbs, but I might be wrong about that figure.

kevinjwalters, 2025-04-02:
A compile option is useful for a minority but won't be of much use to a typical MicroPython programmer. I'd still like this to be increased in size on the V2. Perhaps a compromiste value that's viable in the long-term can be chosen based on the roadmap and guestimate on bloat and additions over time of libraries, SoftDevice and bootloader? As part of the increase in interpreter code size, [mpy support](https://github.com/bbcmicrobit/micropython/issues/524) would be great, at least for code in the file system.

A documentation update with accurate values for V1 and V2 would be useful too.

kevinjwalters, 2025-07-17:
I just noticed this was 32k briefly then got reduced to 24k in https://github.com/microbit-foundation/micropython-microbit-v2/commit/d261e878b39edec5cb2bf73d518ab1effc1265e4 with the comment

> Because the firmware is overlapping with the start of the filesystem, and
there needs to be room to grow the firmware further.

kevinjwalters, 2025-08-07:
@dpgeorge I noticed you reduced the file system space for the V2 MicroPython port from 32k to 24k in https://github.com/microbit-foundation/micropython-microbit-v2/commit/d261e878b39edec5cb2bf73d518ab1effc1265e4 . This leaves it as the same or very similar size to the micro:bit V1. I was surprised by the sized as the nRF52833 (512K) on V2 has twice as much flash as the nRF51822 (256K) on V1.

The performance (4x clock, FPU, etc) and extra memory (4x) give the V2 far more scope for more advanced programs. I've run out of disk space despite minifying my code and library code for one of my [projects](https://www.instructables.com/Making-an-Accurate-Stylish-Clock-With-the-Kitronik/). I was hoping to get the file system enlarged to squeeze some more functionality in. I can't use mpy due to https://github.com/bbcmicrobit/micropython/issues/524

kevinjwalters, 2025-09-18:
The `codal_port` has a dependency on the `nrf` port for the chunk size which is expressed as a 2^n power in `MBFS_LOG_CHUNK_SIZE`. A [comment](https://github.com/micropython/micropython/blob/45ac651d1a2801bccbdc32fddaa9b029ed4ce879/ports/nrf/modules/os/microbitfs.h#L34-L41) discuss the size limitation.

```
#ifndef MBFS_LOG_CHUNK_SIZE
// This property can be tuned to make the filesystem bigger (while keeping
// the max number of blocks). Note that it cannot (currently) be increased
// beyond 8 or uint8_t integers will overflow.
// 2^7 == 128 bytes
// (128-2) bytes/block * 252 blocks = 31752 usable bytes
#define MBFS_LOG_CHUNK_SIZE 7
#endif
```

1. If that's correct then 8 is a viable number and would allow (256-2) * 252 = 64008 bytes presumably with some sort of penalty for small files.
2. Is enlarging the `uint8_t` variables used for chunks to `uint16_t` a better idea?
3. Tangential issue: should codal_port be independent of the `nrf` port?

kevinjwalters, 2025-09-21:
I tried `MBFS_LOG_CHUNK_SIZE` of 8 with 48k file system and it corrupted itself after a replaced one file in the file system, strong suggest that 8 is buggy.

kevinjwalters, 2025-10-01:
I now think issues with file system size changes and chunk size changes are likely to be related to issues described in https://github.com/microbit-foundation/micropython-microbit-v2/issues/231

microbit-carlos, 2025-11-14:
For context, the main reason the filesystem size was left to 20ish KBs was to match the same space available in micro:bit V1.

This is because in mixed classrooms, where students might be using a micro:bit V1 or V2 we wanted to prevent situations where the code would fit only for students using a micro:bit V2.

That being said, ideally we should be able to provide a way for more advance users to expand this filesystem space. A runtime flag could be possible, but would require larger changes, which is why I was thinking a compile time flag could be an easier way to be able to deliver this feature.

@kevinjwalters one question, living `MBFS_LOG_CHUNK_SIZE` to `7` works when the filesystem is expanded the the maximum possible at that configuration (30ish KBs)? Or does that fail as well?